sfeakes/SprinklerD #2: Step by step raspberry pi and HomeKit

toshibochan, 2018-12-01:
Hi thanks for developing greats! I be very happy if you can do step by step installation for raspberry pi and HomeKit. Plus if have YouTube video to. I don’t know much coding so be super super helpful for me. Thank you.

toshibochan, 2018-12-01:
I was able to install the sprinkler but  scheduled not working.
And I’m trying install homebridge but give me error.
Please help.

sfeakes, 2018-12-01:
The schedule is done through linux cron. cron will take a minute or two to sort out it's schedule, so while testing don't schedule it to do something 1 minute in the future. Make sure it's at least 3 mins in the future.
If that doesn't fix it can you post information from /var/log/messages from the time when the schedule was supposed to run.

toshibochan, 2018-12-01:
When I run sudo ./release/install.sh
Give this:
Please enabled cron.d support, if not the calendar will not work
Edit /etc/default/cron and look for the -l option
Created symlink /etc/systemd/system/multi-user.target.wants/sprinklerd.service → /etc/systemd/system/sprinklerd.service.

I did enable cron but I don’t understand what to do with:

Created symlink /etc/systemd/system/multi-user.target.wants/sprinklerd.service → /etc/systemd/system/sprinklerd.service.

toshibochan, 2018-12-01:
I post a question on homebridge-sprinkled page please take look when you have time please.

toshibochan, 2018-12-01:
Schedule is working now thanks

toshibochan, 2018-12-01:
For HomeKit work I need to install mqtt in my pi?

sfeakes, 2018-12-01:
Yes, you do need to install MQTT / Mosquitto.  I've updated the instructions.

toshibochan, 2018-12-01:
Thanks for update! Have on error in there:

Install HomeBridge-SprinklerD

See https://github.com/sfeakes/homebridge-sprinklerd for package content

sudo npm install -g homebridge-aqualinkd

Is not aqualinkd.

toshibochan, 2018-12-01:
After install mosquito how to setup mosquito and start the mosquito?

toshibochan, 2018-12-01:
When I start sprinklerd give me this:

Notice: Starting MQTT client to trident:1883
Info:   Connected to mqtt trident:1883 with id of: sprinklerd_b827eb94711c
Warning:MQTT Connection closed
Notice: MQTT client stopped

This normal?

toshibochan, 2018-12-02:
this message just show and show and never stop:

Notice: Starting MQTT client to trident:1883
Info:   Connected to mqtt trident:1883 with id of: sprinklerd_b827eb94711c
Warning:MQTT Connection closed
Notice: MQTT client stopped

I have to open port 1883 on the router?

sfeakes, 2018-12-02:
No, your router is to the outside world, you shouldn’t open that port since it’s all local.
Arr, you are trying to connect to mosquitto on a machine named trident, that should probably be localhost.  Look at the sprinklerd.conf file and change the MQTT settings to the machine running mosquitto.

toshibochan, 2018-12-02:
You can share your sprinklerd.config file?

toshibochan, 2018-12-02:
I was able to connect to mosquitto but now new error:
Debug:  MQTT: topic sprinklerd/zone1/set 1
Debug:  MQTT: pt2 zone1/set1ngduration0@
Debug:  Request to turn zone 1 on for 0 min requesttype 1
Info:   Stopping!
 And crash

toshibochan, 2018-12-02:
[SPRINKLERD]
PORT=105
NAME=My Sprinklers
DOCUMENTROOT = /var/www/sprinklerd 
CACHE = /var/cache/sprinklerd.cache
#The log level. [DEBUG, INFO, NOTICE, WARNING, ERROR]
LOG_LEVEL = DEBUG
LOG_LEVEL = NOTICE 

mqtt stuff
MQTT_ADDRESS = 10.0.1.107:1883
#MQTT_USER = username
#MQTT_PASSWD = password
MQT_TOPIC = sprinklerd
#MQTT_DZ_PUB_TOPIC = domoticz/in
#MQTT_DZ_SUB_TOPIC = domoticz/out

#DZIDX_CALENDAR = 197
#DZIDX_24HDELAY = 198
#DZIDX_ALL_ZONES = 199
#DZIDX_RAINSENSOR = 48

# Options for the below ZONE and GPIO configuration
#
# LOW        0
# HIGH       1
#
# PUD_OFF      0
# PUD_DOWN     1
# PUD_UP       2

#NAME           = name of zone
#GPIO_PIN       = GPIO Pin # This is WiringPi Pin#, not Raspberry Pi board pin#.
#WPI_PIN        = use instead of GPIO_PIN if compiled with USE_WIRINGPI flag
#GPIO_PULL_UPDN = setup pull up pull down resistor  PUD_OFF|PUD_DOWN|PUD_UP
#GPIO_ON_STATE  = State GPIO reads when relay for zone is on. HIGH or LOW  1 or 0
#DOMOTICZ_IDX   = Domoticz IDX 0 or remove entry if you don;t use Domoticz (only if you use domoticz)
#MASTER_VALVE   = YES=1 NO=0  // turn on with any zone

# Don't use ZONE:0 for anything other than master valve, if you don't have a master valve simply delete it and start from ZONE:1
[ZONE]
[ZONE:0]
NAME=Master Valve
MASTER_VALVE=0
GPIO_PIN=17
WPI_PIN=0
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0

[ZONE:1]
NAME=Island
DEFAULT_RUNTIME=10
GPIO_PIN=18
WPI_PIN=1
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=200

[ZONE:2]
NAME=Driveway
DEFAULT_RUNTIME=10
GPIO_PIN=27
WPI_PIN=2
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=201

[ZONE:3]
NAME=Diningroom Flowerbeds
DEFAULT_RUNTIME=10
GPIO_PIN=22
WPI_PIN=3
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=202

[ZONE:4]
NAME=Front Flowerbeds
DEFAULT_RUNTIME=10
GPIO_PIN=23
WPI_PIN=4
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=203

[ZONE:5]
NAME=Backgarden Left
DEFAULT_RUNTIME=10
GPIO_PIN=24
WPI_PIN=5
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=204

[ZONE:6]
NAME=Backgarden Right
DEFAULT_RUNTIME=10
GPIO_PIN=25
WPI_PIN=6
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=205

[ZONE:7]
NAME=Garage Flowerbeds
DEFAULT_RUNTIME=10
GPIO_PIN=5
WPI_PIN=21
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=206

[ZONE:8]
NAME=Golfcart path
DEFAULT_RUNTIME=10
GPIO_PIN=6
WPI_PIN=22
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=207

toshibochan, 2018-12-02:
>
[SPRINKLERD]
PORT=105
NAME=My Sprinklers
DOCUMENTROOT = /var/www/sprinklerd 
CACHE = /var/cache/sprinklerd.cache
#The log level. [DEBUG, INFO, NOTICE, WARNING, ERROR]
LOG_LEVEL = DEBUG
LOG_LEVEL = NOTICE 

mqtt stuff
MQTT_ADDRESS = 10.0.1.107:1883
#MQTT_USER = username
#MQTT_PASSWD = password
MQT_TOPIC = sprinklerd
#MQTT_DZ_PUB_TOPIC = domoticz/in
#MQTT_DZ_SUB_TOPIC = domoticz/out

#DZIDX_CALENDAR = 197
#DZIDX_24HDELAY = 198
#DZIDX_ALL_ZONES = 199
#DZIDX_RAINSENSOR = 48

# Options for the below ZONE and GPIO configuration
#
# LOW        0
# HIGH       1
#
# PUD_OFF      0
# PUD_DOWN     1
# PUD_UP       2

#NAME           = name of zone
#GPIO_PIN       = GPIO Pin # This is WiringPi Pin#, not Raspberry Pi board pin#.
#WPI_PIN        = use instead of GPIO_PIN if compiled with USE_WIRINGPI flag
#GPIO_PULL_UPDN = setup pull up pull down resistor  PUD_OFF|PUD_DOWN|PUD_UP
#GPIO_ON_STATE  = State GPIO reads when relay for zone is on. HIGH or LOW  1 or 0
#DOMOTICZ_IDX   = Domoticz IDX 0 or remove entry if you don;t use Domoticz (only if you use domoticz)
#MASTER_VALVE   = YES=1 NO=0  // turn on with any zone

# Don't use ZONE:0 for anything other than master valve, if you don't have a master valve simply delete it and start from ZONE:1
[ZONE]
[ZONE:0]
NAME=Master Valve
MASTER_VALVE=0
GPIO_PIN=17
WPI_PIN=0
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0

[ZONE:1]
NAME=Island
DEFAULT_RUNTIME=10
GPIO_PIN=18
WPI_PIN=1
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=200

[ZONE:2]
NAME=Driveway
DEFAULT_RUNTIME=10
GPIO_PIN=27
WPI_PIN=2
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=201

[ZONE:3]
NAME=Diningroom Flowerbeds
DEFAULT_RUNTIME=10
GPIO_PIN=22
WPI_PIN=3
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=202

[ZONE:4]
NAME=Front Flowerbeds
DEFAULT_RUNTIME=10
GPIO_PIN=23
WPI_PIN=4
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=203

[ZONE:5]
NAME=Backgarden Left
DEFAULT_RUNTIME=10
GPIO_PIN=24
WPI_PIN=5
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=204

[ZONE:6]
NAME=Backgarden Right
DEFAULT_RUNTIME=10
GPIO_PIN=25
WPI_PIN=6
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=205

[ZONE:7]
NAME=Garage Flowerbeds
DEFAULT_RUNTIME=10
GPIO_PIN=5
WPI_PIN=21
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=206

[ZONE:8]
NAME=Golfcart path
DEFAULT_RUNTIME=10
GPIO_PIN=6
WPI_PIN=22
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
DOMOTICZ_IDX=207
<

toshibochan, 2018-12-02:
Debug:  Config mqtt_topic 'sprinklerd'
Debug:  Config mqtt_dz_pub_topic ''
Debug:  Config mqtt_dz_sub_topic ''
Info:   Name = My Sprinklers
Info:   Port = 105
Info:   Docroot = /var/www/sprinklerd
Debug:  Found 8 ZONES
Debug:  ZONE = 0
Debug:  Zone Config        : Master Valve
                      PIN : 17
         Set pull up/down : 1
                 ON state : 0
             Master valve : 0
             Domoticz IDX : -1
Debug:  ZONE = 1
Debug:  Zone Config        : Island
                      PIN : 18
         Set pull up/down : 1
                 ON state : 0
             Master valve : 111
             Domoticz IDX : 200
Debug:  ZONE = 2
Debug:  Zone Config        : Driveway
                      PIN : 27
         Set pull up/down : 1
                 ON state : 0
             Master valve : 84
             Domoticz IDX : 201
Debug:  ZONE = 3
Debug:  Zone Config        : Diningroom Flowerbeds
                      PIN : 22
         Set pull up/down : 1
                 ON state : 0
             Master valve : 97
             Domoticz IDX : 202
Debug:  ZONE = 4
Debug:  Zone Config        : Front Flowerbeds
                      PIN : 23
         Set pull up/down : 1
                 ON state : 0
             Master valve : 73
             Domoticz IDX : 203
Debug:  ZONE = 5
Debug:  Zone Config        : Backgarden Left
                      PIN : 24
         Set pull up/down : 1
                 ON state : 0
             Master valve : 100
             Domoticz IDX : 204
Debug:  ZONE = 6
Debug:  Zone Config        : Backgarden Right
                      PIN : 25
         Set pull up/down : 1
                 ON state : 0
             Master valve : 10
             Domoticz IDX : 205
Debug:  ZONE = 7
Debug:  Zone Config        : Garage Flowerbeds
                      PIN : 5
         Set pull up/down : 1
                 ON state : 0
             Master valve : 35
             Domoticz IDX : 206
Debug:  ZONE = 8
Debug:  Zone Config        : Golfcart path
                      PIN : 6
         Set pull up/down : 1
                 ON state : 0
             Master valve : 79
             Domoticz IDX : 207
[... 237 lines cut ...]
Debug:  Message publishing acknowledged (msg_id: 30)
Debug:  MQTT: received (msg_id: 30), looks like my own message, ignoring
Debug:  Message publishing acknowledged (msg_id: 31)
Debug:  MQTT: received (msg_id: 31), looks like my own message, ignoring
Debug:  action_web_request / type=homebridge
Debug:  Request type homebridge
Debug:  MQTT: topic sprinklerd/zone1/set 1
Debug:  MQTT: pt2 zone1/set1P/1.1
host: 10.0.1.107:105
accept: application/json
Connection: close

rd/statusOFFsOFFleallzones0@
Debug:  Request to turn zone 1 on for 0 min requesttype 1
Info:   Stopping!
Debug:  Turning off Zone 0. Setting pin 17 to state 1
Debug:  Turning off Zone 1. Setting pin 18 to state 1
Debug:  Turning off Zone 2. Setting pin 27 to state 1
Debug:  Turning off Zone 3. Setting pin 22 to state 1
Debug:  Turning off Zone 4. Setting pin 23 to state 1
Debug:  Turning off Zone 5. Setting pin 24 to state 1
Debug:  Turning off Zone 6. Setting pin 25 to state 1
Debug:  Turning off Zone 7. Setting pin 5 to state 1
Debug:  Turning off Zone 8. Setting pin 6 to state 1
Info:   Wrote cache

toshibochan, 2018-12-02:
In the web interface all work good.

What is working on homekit:
Cycle all zone ON and OFF
24h Rain Delay ON and OFF
Run on calendar schedule ON and OFF
When switch ON cycle all zone the relay work okay

What not working on homekit:
All other sprinkler button not working.
Every time i try to switch ON sprinkler the sprinklerd crashes.

sfeakes, 2018-12-02:
Thanks for this.  There was a bug in the code.  I've posted a fix, so if you update to the latest SprinklerD version it should be fixed. (No need to update homekit-sprinklerd).

toshibochan, 2018-12-02:
Yay it works!!!!! Thank you! Thank you! Thank you!
Just one request.
On HomeKit now only can set time 5 min 10 min 15 min...... is possible to change the code for every 1 min?

toshibochan, 2018-12-02:
I see. Let me know if you find way to do every min.
One more think I was looking for do schedule on HomeKit app but I wasn’t able to select any sprinkler.

sfeakes, 2018-12-02:
I posted information in the other thread.  It's a limitation of the apple HomeKit app.

https://github.com/sfeakes/SprinklerD/issues/1

sfeakes, 2018-12-03:
Looks like scheduling on HomeKit app, is again limitation of HomeKit app.

toshibochan, 2018-12-03:
Apple have to stop all this limitations! I test with eve app and work

sfeakes, 2018-12-03:
I agree.  Other apps have a lot more control over HomeKit accessories than the Apple app does.  But the Apple app is still the most user friendly / easy to use.

closed 2018-12-03 by toshibochan