sfeakes/SprinklerD #3: MASTER_VALVE=0

toshibochan, 2018-12-03:
I set MASTER_VALVE=0 but the master relay still turning ON. This is normal?

sfeakes, 2018-12-03:
If you have zone 0 configured, that's seen as a master valve and will turn on with any other zone.  So if you don't have a master valve, start at zone 1.
Looking at it, the master_valve setting seems redundant.

sfeakes, 2018-12-03:
I just checked the code. I had removed the master_valve part of the zone configuration, just forgot to take it out of the configuration file.  So that MASTER_VALVE=0 or 1 has no meaning, I've pulled it from the config file.

toshibochan, 2018-12-03:
After the update the schedule not working anymore. I look in /etc/default/cron and EXTRA_OPTS=‘-l’ is enable.

toshibochan, 2018-12-03:
When I delete the zone 0 the web not working

toshibochan, 2018-12-03:
I reinstall all and the schedule is working now but when I delete zone 0 from conf file the web no show the other zones

sfeakes, 2018-12-03:
There was a bug that causes some errors when zone 0 was not configured, I've fixed that and posted an update.

toshibochan, 2018-12-03:
Thank you! And sorry for bothering you all the time! And sorry if my English is bad! I’m still learning.

toshibochan, 2018-12-03:
I’m trying to set 2 zones to 0 min but keep going back to 10 min. Have any way to set to 0 min?

toshibochan, 2018-12-03:
When you update the codes on github how you update your raspberry pi?

sfeakes, 2018-12-03:
Go to the directory where you have SprinklerD and run the following 
‘’’
git pull
sudo ./release/install.sh
‘’’

toshibochan, 2018-12-03:
Delete master valve now is okay.
Now only last thing is 0 min.

sfeakes, 2018-12-03:
For default run time, 0 mins is not really valid, so thats why it's rejecting it and resetting to 1.
For the calendar schedule 0 is a valid runtime, so you should be able to set that to 0.

toshibochan, 2018-12-04:
How to uninstall all?
I want uninstall all and install from scratch again.

sfeakes, 2018-12-04:
if you download the latest version, I've added an uninstall option to the install script.
'''./release/install.sh uninstall'''

if you want to do it manually, remove these files.
'''
/usr/local/bin/sprinklerd
/etc/sprinklerd.conf
/etc/systemd/system/sprinklerd.service
/etc/default/sprinklerd
/var/www/sprinklerd/<remove all sub directories>
/etc/avahi/services/sprinklerd.service
/etc/cron.d/sprinklerd
/var/cache/sprinklerd.cache
'''

toshibochan, 2019-01-08:
Hope one day you can find how to make “run now option” to set to “0 minute”. Because I have one zone in the room for in case of fire. So I don’t want this zone to run.

sfeakes, 2019-01-09:
The latest update supports setting default run time to 0.

toshibochan, 2019-01-09:
Super thanks! work great!

closed 2019-01-09 by toshibochan