sfeakes/SprinklerD #4: Segfault when enabling MQTT

cmlpreston, 2019-01-05:
HI, I think I'm getting a segfault when MQTT is enabled. 

The intHandler() is getting called at line 356 of sprinkler.c with a SEGFAULT signal. This only happens when I enable MQTT as follows:

 10 # mqtt stuff (address and topic are mandatory if you want to use MQTT)
 11 MQTT_ADDRESS = 192.168.1.19:1883
 12 MQTT_USER = pi
 13 MQTT_PASSWD = pi
 14 MQT_TOPIC = sprinklerd

Any help appreciated. I'm happy to debug further.

cmlpreston, 2019-01-05:
I've now narrowed the segfault to this line in publish_zone_mqtt()

sprintf(mqtt_msg, "%s", (digitalRead(gpiopin->pin) == gpiopin->on_state ? MQTT_ON : MQTT_OFF));

I'll investigate more later.

cmlpreston, 2019-01-05:
I've switched to the WiringPi version. Works great!

sfeakes, 2019-01-05:
Great piece of debugging.  WiringPI has far more sanity checks than my GPIO implementation, but it's huge by comparison.  Still be interested in why my GPIO implementation was crashing on your system.  What version of OS and more importantly Board / SBC are you using?

cmlpreston, 2019-01-07:
Hi, here's what sprinklerd reports for board id:

piBoardId: Revision string: Revision#011: a02082

and

$ uname -a
Linux pi3 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

and 

 $ cat /proc/device-tree/model 
Raspberry Pi 3 Model B Rev 1.2

HTH!

sfeakes, 2019-01-07:
Thanks so much, think I am not reading your board as Pi3, and that’s what would cause the crash as I’m reading / writing to the wrong memory address for the GPIO.  Can you please run ‘cat /proc/cpuinfo‘ and post the output.  Thanks again.

cmlpreston, 2019-01-07:
No problem. Here's the output.

$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000003567a948

sfeakes, 2019-01-09:
I push an update, few small changes, not sure I fixed this, but if you could get the latest version and try the pre-built binary be interesting to know if it fixed it.  I wouldn't overwrite your existing install just yet.  So just to a git clone to a different/new directory, stop sprinklerd and run
```sudo ./release/sprinklerd -d -c <existing config file>```

See if that works for you, if not, can you post the log. (please set debug mode)

Cheers

cmlpreston, 2019-01-09:
Looks like there's a problem reading the zones from the conf file with 1.0e.

Previous (WPi) version 1.0d output:

Debug:  Config mqtt_topic 'sprinklerd'
Debug:  Config mqtt_dz_pub_topic ''
Debug:  Config mqtt_dz_sub_topic ''
Info:   Name = XXX Sprinklers
Info:   Port = 8080
Info:   Docroot = /var/www/sprinklerd
Debug:  Found 2 ZONES
Debug:  No Information for ZONE = 0
Debug:  ZONE = 1
Debug:  Zone Config        : Fence side
                      PIN : 9
         Set pull up/down : 0
                 ON state : 1
             Domoticz IDX : -1
Debug:  ZONE = 2
Debug:  Zone Config        : House side
                      PIN : 8
         Set pull up/down : 0
                 ON state : 1
             Domoticz IDX : -1
Notice: Starting ./sprinklerd version 1.0d
Debug:  regexp no match (1)
Debug:  Read from cron Day 0 | Time 20:0 | Zone 1 | Runtime 15
...

New version (1.0e):

Debug:  Config mqtt_topic 'sprinklerd'
Debug:  Config mqtt_dz_pub_topic ''
Debug:  Config mqtt_dz_sub_topic ''
Info:   Name = XXX Sprinklers
Info:   Port = 8080
Info:   Docroot = /var/www/sprinklerd
Debug:  Found 0 ZONES
ERROR no config zones set
Error:   no config zones set

sfeakes, 2019-01-09:
Thanks so much for trying,  

In your config, you have
GPIO_PIN=3
GPIO_PIN=2

for your two zones.  Just out of interest, when you have time, can you try different pins, eg
GPIO_PIN=17
GPIO_PIN=18

Obviously the relays won't click over, but that's not important.  I'd like to see if this is a memory address issue or some access issue.  If same result, it's probably former, if works probably latter.

Thanks again for all your work.

sfeakes, 2019-01-09:
BTW, I'm sure it will work if you uncomment `#define GPIO_SYSFS_MODE` in sd_GPIO.h, but you are better off using wiringPi than my GPIO implementation in sysfs mode.  Well I say it will work, but you might get a different error as I haven't tested sysfs mode in a while...:)

sfeakes, 2019-01-09:
Never mind, I can replicate the issue.  I found a Pi same as yours, used your exact config and got the same error.   Hope to post a fix soon.

cmlpreston, 2019-01-09:
Excellent, looks good to me too, although I'm at work and can't tell if the relays are operating! Do you recommend switching back to your GPIO code rather than WiringPi?

sfeakes, 2019-01-09:
Switching back is up to you, it’ll save you having to compile each time you upgrade, but that’s about it.  If you are using wiringpi for anything else on that particular pi or using some funky relay board that necessitates features of wiring pi, then I’d recommend keeping it compiled with wiring pi.
I wrote my own GPIO simply to keep everything as small as possible and to save people the hassle of installing wiringpi and keeping 2 things up to date and compiled together. But speed / implementation wise they are identical.

When I get round to adding features for reading GPIO sensors & switches in the future, then I’d recommend using my implementation if you use want to use those.

closed 2020-09-27 by sfeakes