sfeakes/SprinklerD #5: Request future

toshibochan, 2019-01-08:
I want be able to have push buttons to start watering manually. One button for each zone, push onence start watering,push one more time stop watering.

You think this is possible?

Thank you!  :)

sfeakes, 2019-01-09:
This shouldn't be hard.  you can use lots of software to read button press from the GPIO, and when it detects a press, call the following URL
localhost?type=zone&zone=2&state=flip
That will toggle the zone state.

I will be adding the code to read the button press within SprinklerD in a future release, month or two. So soon you should be able to do it all within SprinklerD

toshibochan, 2019-01-11:
Thank you! Be nice future if have this option! This way the gardener can turn ON and OFF manually

toshibochan, 2021-02-15:
@sfeakes thank you for fixing the zone 1 issue. when you have time, you can make to change the zone name in the web interface? and in the schedule show the name off the zone?