sfeakes/SprinklerD #6: need help on schedule

toshibochan, 2019-03-17:
I did I clean install this week in 2 raspberry pizero.
Turn ON and OFF manually is working OK, but schedule not working in both PI.
I did:
sudo nano /etc/default/cron
#EXTRA_OPTS='-l' change to EXTRA_OPTS='-l'

toshibochan, 2019-03-17:
EXTRA_OPTS='-l' or is EXTRA_OPTS="-l"?

toshibochan, 2019-03-17:
raspberry pi zero
raspberry pi zero W

latest Raspbian "Raspbian Stretch with desktop and recommended software" 2018-11-13
node -v is v10.15..1
wiringpi -v2.46

toshibochan, 2019-03-17:
I see on Debug and when I change schedule showing "updating cron" but schedule not working.

toshibochan, 2019-03-17:
strange is I did new test on raspberry pi 3 B plus and is working only in 2 pi zero not working

toshibochan, 2019-03-28:
on pi3 I do this
/usr/bin/curl -s -o /dev/null 'localhost?type=cron&zone=1&runtime=1&state=on'
 and the relay turn on but on PiZero nothing happen.

please help

sfeakes, 2019-03-28:
Sorry I completely missed this thread.

Pi zero and Pi3 use the same GPIO memory address, so that's odd.  If you are sure you have the correct GPIO pin configuration (in the config file) and connections to relay board (hardware), it may be worth compiling with WiringPi rather than the builtin GPIO control.  Info is on the main page on how to do that.

toshibochan, 2019-03-28:
I did clean install on pi zero and pi 3 and did same test. And all work in pi 3. The pi zero only schedule don’t work. I can turn on and off manually but not schedule.

toshibochan, 2019-03-28:
I see /etc/cron.d/sprinklerd update okay but can’t control on schedule

sfeakes, 2019-03-28:
Did you check cron.d is enabled and working? that's usually the first thing to check.  Looks like you did from messages above, just making sure you did it on both pi3 and pi zero.

Next try making curl not silent, see if you get an error
```
/usr/bin/curl 'localhost?type=cron&zone=1&runtime=1&state=on'
```

toshibochan, 2019-03-28:
I did activate cron -l same on pi zero and pi 3
I did this /usr/bin/curl 'localhost?type=cron&zone=1&runtime=1&state=on' on pi 3 and work but on pi zero nothing happens

toshibochan, 2019-03-28:
on pizero come back 
pi@PiZero:~ $ /usr/bin/curl 'localhost?type=cron&zone=1&runtime=1&state=on'
curl: (7) Failed to connect to localhost port 80: Connection refused

I see what happen. is because I change the port address. I did put back to port 80 and now working good.

sorry for my mistake

and thank you for helping me

toshibochan, 2019-03-28:
sorry no work
/usr/bin/curl 'localhost?type=cron&zone=1&runtime=1&state=on'

/usr/bin/curl 'localhost?type=cron&zone=1&runtime=1&state=on'
{ "title" : "My Sprinklers","calendar" : "on", "24hdelay" : "off", "allz" : "off", "zones" : "5", "24hdelay-offtime" : 0, "status" : "Active: Zone 1 : time left 01:00", "raindelaychance" : "0", "raindelaytotal1" : "0.0", "raindelaytotal2" : "0.0", "todaychanceofrain" : "0", "todayraintotal" : "0.00", "z1" : "on" , "z2" : "off" , "z3" : "off" , "z4" : "off" , "z5" : "off" }

this work on port 80 but the schedule still not working.

sfeakes, 2019-03-28:
If you are not using port 80, then I just realized there is a bug. The schedule always expects port 80.

sfeakes, 2020-09-27:
Fixed this long time ago, forgot to update this thread, so closing

closed 2020-09-27 by sfeakes