sfeakes/SprinklerD #8: Feature request

swapper82, 2019-06-05:
Would be perfect if the software allowed no usage of Rpi GPIO and only Domoticz IDX devices as switches (i guess it only supports triggering from Domoticz not to domoticz) . I have all valves controlled by Domoticz but this applications integrations against homekit is so much cleaner since it uses "valves"

swapper82, 2019-06-05:
Found https://github.com/goedh452/homebridge-http-sprinkler that solves the issue i have minus the nice webinterface though.

sfeakes, 2019-06-05:
Glad you found a solution.  But, this is bi-directional with both Domoticz & Homekit, so it would do exactly what you want. But, it does used the GPIO memory address for the valve state, so just don;t connect a relayboard to the GPIO pins and it would do what you want (as long as you didn;t use the GPIO# for something else)

closed 2019-06-05 by sfeakes