sfeakes/SprinklerD #10: Homebridge-sprinklerd plug-in not loading correctly

woodgieboodgie, 2019-07-16:
We have installed the homebridge-sprinklerd plug in successfully and added the platform into config.json.

When we run homebridge it says that the platform sprinklerd is not linked to any plug in.

sfeakes, 2019-07-16:
This means that the homebridge-sprinklerd did not register/install with the homebridge instance you are running.  Debugging this will be very dependent on the homebridge install you are using.
Are you sure you don;t have multiple homebridge installs on the same machine?  If you are, can you list / describe how & where you installed homebridge.

woodgieboodgie, 2019-07-17:
I believe/hope that only one instance of Homebridge is installed.  There are several platforms and accessories working.  I followed the installation instructions for Homebridge at https://github.com/nfarina/homebridge.  I think it was installed in root(?) directory and can be found at /.homebridge.  I could see SprinklerD via the browser too

I have the following info for you if it helps, pls let me know what else you need
Curl 7.52.1-5
Node 8.16.0
Homebridge 0.4.50
uName Linux rpi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

woodgieboodgie, 2019-07-17:
When trying to perform the following: 'sudo npm install -g homebridge-sprinklerd' it returned an error.  typed sudo bash and then repeated the command and it appears to have worked.  icons appear within homebridge and I can turn the zones on and off -- for a few mins.  Then it stops working.  Buttons in homebridge illuminate on/off but are not controlling the 5v 8-channel board any more. Also, it appears to work on the zones titles form the original sprinklerd.conf.  if I changed those it stopped working, so I changed them back.  Please help as I must be close now.

woodgieboodgie, 2019-07-17:
[image]

I noticed the sprinklerd.conf created the accessories, but when their names were changed in the file, and the system rebooted, the same ones appeared (per image above)

sfeakes, 2019-07-18:
You should do the following in this order.

Stop Homebridge and sprinklerd
delete the Homebridge cache
Configure sprinklerd correctly.
Start sprinklerd
Check sprinklerd with the web ui. (Reconfigure as necessary, stop & restart as needed each time you change the config)
Start HomeKit.

That should reset all the devices.  (They should automatically delete from HomeKit and be recreated).

woodgieboodgie, 2019-07-18:
OK.  Pardon my stupid question.. but does sprinklerd not stop when homebridge is stopped?  What is the command to stop it separately if so?  Also, when you say cache, do you mean all files in /accessories, or /persist?  Deleting them both means i will have to set them up on all devices again.. doesnt it?

sfeakes, 2019-07-18:
No, sprinklerd is its own daemon, sprinklerd-Homebridge is a plug-in for Homebridge to allow communication between sprinklerd and HomeKit.
The below will stop sprinklerd 

sudo systemctl stop sprinklerd

Replace stop with start to start it.

woodgieboodgie, 2019-07-18:
Which files do I need to delete specifically to remove cache?  Do I really need to remove all files in the /accessories, or /persist?

sfeakes, 2019-07-18:
All files in both directories.   That will delete all Homebridge accessories though, so if you are using Homebridge for something other than sprinklerd, it will reset those as well.

woodgieboodgie, 2019-07-21:
All files in both directories removed.  Started pi and homebridge and correct values for zones have come across to the app.

closed 2019-07-21 by woodgieboodgie