sfeakes/SprinklerD #11: Scheduling assistance

woodgieboodgie, 2019-07-21:
Hi.  The schedule is accessible via the web browser and when i enter start times for each zone, the start time passes without the program starting the zones.  fyi, i can see the zones and manually activate them via the browser.

sfeakes, 2019-07-21:
cron.d is probably disabled and you missed the warning message from the install script.

Please look in /etc/default/cron and make sure you have the following set
```EXTRA_OPTS='-l'```
(make sure no `#` at the start of the line)

That wil enable cron.d, which is what sprinklerd uses for the scheduler.

Also remember that you need to set the start time at least 2~4 mins from "now" for the first run, this will allow cron.d to schedule the job.  IE, if current time is 12:10 and you schedule for 12:11 it won't come on for 24 hours.  But 12:15 will come on in 5 mins. (not the best explanation, but hope you understand)

woodgieboodgie, 2019-07-21:
I had removed the ## already and setup the schedule to start zone 1 in 5 mins. The time passed and the zone did not trigger.

sfeakes, 2019-07-21:
If you list the file `/etc/cron.d/sprinklerd` you'll see a bunch of commands like
```
45 7 * * 5 root /usr/bin/curl -s -o /dev/null 'localhost?type=cron&zone=8&runtime=5&state=on'
```

If you run one of those commands from the /usr part as root, can you post what happens
ie.
```
sudo /usr/bin/curl -s -o /dev/null 'localhost?type=cron&zone=8&runtime=5&state=on'
```

Also, can you post your `/etc/sprinklerd.conf` file, and `/etc/cron.d/sprinklerd`, from those I should be able to tell what's going on.

woodgieboodgie, 2019-07-21:
/etc/cron.d/sprinklerd =

#***** AUTO GENERATED DO NOT EDIT *****
0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost?type=sensor&sensor=chanceofrain&value=0'
0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost?type=sensor&sensor=raintotal&value=0'
#***** AUTO GENERATED DO NOT EDIT *****

sprinklerd.conf =

[SPRINKLERD]
PORT=80
NAME=My Sprinklers
DOCUMENTROOT = /var/www/sprinklerd 
CACHE = /var/cache/sprinklerd.cache
# The log level. [DEBUG, INFO, NOTICE, WARNING, ERROR]
#LOG_LEVEL = DEBUG
LOG_LEVEL = NOTICE 

# mqtt stuff (address and topic are mandatory if you want to use MQTT)
MQTT_ADDRESS = localhost:1883
#MQTT_USER = someusername    
#MQTT_PASSWD = somepassword
MQT_TOPIC = sprinklerd

# if you are using domoticz and MQTT uncomment
#MQTT_DZ_PUB_TOPIC = domoticz/in
#MQTT_DZ_SUB_TOPIC = domoticz/out

#DZIDX_CALENDAR = 197
#DZIDX_24HDELAY = 198
#DZIDX_ALL_ZONES = 199
#DZIDX_RAINSENSOR = 48

# Options for the below ZONE and GPIO configuration
#
# LOW        0
# HIGH       1
#
# PUD_OFF      0
# PUD_DOWN     1
# PUD_UP       2

#NAME           = name of zone
#GPIO_PIN       = GPIO # This is the GPIO# not the pin#, so (17 = GPIO17 = Pin 11) or another example (5 = GPIO5 = Pin 29)
#WPI_PIN        = use instead of GPIO_PIN if compiled with USE_WIRINGPI flag, This is WiringPi Pin#, not Raspberry Pi board pin#
#GPIO_PULL_UPDN = setup pull up pull down resistor  PUD_OFF|PUD_DOWN|PUD_UP
#GPIO_ON_STATE  = State GPIO reads when relay for zone is on. HIGH or LOW  1 or 0
#DOMOTICZ_IDX   = Domoticz IDX 0 or remove entry if you don;t use Domoticz (only if you use domoticz)

# Don't use ZONE:0 for anything other than master valve, if you don't have a master valve simply delete it and start from ZONE:1
[ZONE]
[ZONE:0]
NAME=Master Valve
GPIO_PIN=17
WPI_PIN=0
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0

[ZONE:1]
NAME=Front Garden
DEFAULT_RUNTIME=10
GPIO_PIN=18
WPI_PIN=1
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=200

[ZONE:2]
NAME=Back Garden
DEFAULT_RUNTIME=10
GPIO_PIN=27
WPI_PIN=2
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=201

[ZONE:3]
NAME=Lawn West
DEFAULT_RUNTIME=10
GPIO_PIN=22
WPI_PIN=3
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=202

[ZONE:4]
NAME=Lawn East
DEFAULT_RUNTIME=10
GPIO_PIN=23
WPI_PIN=4
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=203

[ZONE:5]
NAME=Lawn South
DEFAULT_RUNTIME=10
GPIO_PIN=24
WPI_PIN=5
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=204

[ZONE:6]
NAME=Pool Top Up
DEFAULT_RUNTIME=10
GPIO_PIN=25
WPI_PIN=6
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=205

[ZONE:7]
NAME=Storm Water
DEFAULT_RUNTIME=10
GPIO_PIN=5
WPI_PIN=21
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=206

sfeakes, 2019-07-21:
There aren't any schedules listed in /etc/cron.d/sprinklerd, the two that are listed are to reset the rain sensor each day.

Not sure what you're doing / not doing but play with the scheduler UI a bit (different browser maybe) and see if you get anything written to /etc/cron.d/sprinklerd, you're looking for a line similar to this
```
45 7 * * 5 root /usr/bin/curl -s -o /dev/null 'localhost?type=cron&zone=8&runtime=5&state=on'
```

woodgieboodgie, 2019-07-23:
Just confirming there is no 'apply' button in the scheduler and that once a number (greater than zero) is written into any zone at that point it is committed.
The browser I was using was Safari on the iphone 6s.  Will try a computer based browser next.

toshibochan, 2019-07-23:
Did you edit /etc/default/cron?
sudo nano /etc/default/cron
# EXTRA_OPTS='-l'
Delete #
EXTRA_OPTS='-l'

woodgieboodgie, 2019-07-23:
Thanks Toshibochan, that has already been done.  Trying to ascertain whether the times i enter into the schedule are in fact getting scheduled as my output above indicates they are not.

woodgieboodgie, 2019-07-23:
**sudo /usr/bin/curl -s -o /dev/null 'localhost?type=cron&zone=1&runtime=2&state=on'** 

the command above got zone one working for the desired 1 minute.  why can the schedule not do the same?  i am using port 80

sfeakes, 2019-07-23:
> 
> 
> Just confirming there is no 'apply' button in the scheduler and that once a number (greater than zero) is written into any zone at that point it is committed.
> The browser I was using was Safari on the iphone 6s. Will try a computer based browser next.

There is no "apply button", but you need to tab-off / move to a different cell for the value in the browser to be sent back to sprinklerd.  Then when sprinklerd see's the update made to the schedule, it will wait ~30 seconds for any other updates to come, then write to the file `/etc/cron.d/sprinklerd`.

Easy way to check if the value was sent is to refresh the browser and see if the new setting is listed in the schedule

> 
> 
> **sudo /usr/bin/curl -s -o /dev/null 'localhost?type=cron&zone=1&runtime=2&state=on'**
> 
> the command above got zone one working for the desired 1 minute. why can the schedule not do the same? i am using port 80

Has that line been written to the file  `/etc/cron.d/sprinklerd` ?  If not, that's why it doesn't run the schedule.

> You may also want to check /var/log/messages for any sprinklerd errors.

Have you looked for any errors in the log?  If you have and nothing came up, run sprinklerd manually in debug mode and look at the output for errors (or post output)

stop sprinklerd `sudo systemctl stop sprinklerd`
run manually `sudo sprinklerd -v -d -c /etc/sprinklerd.conf`

woodgieboodgie, 2019-07-24:
how do i place sprinklerd into verbose logging mode?

woodgieboodgie, 2019-07-24:
edit: i successfully ran it manually and could see the data in the window.  however i want to be able to capture and review the log offline

sfeakes, 2019-07-24:
As above -v sets verbose (debug) mode when running from command line.
If you run it as a daemon, Look in the config file for log debug and go to where ever you have syslog logging to for the output (usually /var/log/messages), But you should be able to scroll back in the terminal where you started sprinklerd (with the command in my last post) to see any errors.  Or simply run the command in my previous post and pipe to a file if you can’t scroll back in the terminal. Ie sudo sprinklerd -v -d -c /etc/sprinklerd.conf > sprinklerd.log

woodgieboodgie, 2019-08-09:
ran manually sudo sprinklerd -v -d -c /etc/sprinklerd.conf and scrolled up the screen.
could not see any errors sticking out, the only one that raised my interest was a message suggesting a message had come from itself and therefore was being ignored.

do the four lines in the sprinklerd.conf relating to mqtt need to be active, ie: no #10 i dont have a setting in username and password, and the ip points to the localhost, is that correct?

i can also confirm when i change times in the schedule, they get written to /etc/cron.d/sprinklerd.

woodgieboodgie, 2019-08-09:
furthermore, i set the schedule to run in 5 minutes time, giving it ample time to do what it needs ahead of running.  when the time arrives, the schedule still doesnt kick in and start the zones.
 are you able to provide step by step fault diagnosis?

sfeakes, 2019-08-09:
Before from your posts lines (the schedule) were not getting written to /etc/corn.d/sprinklerd. 2 posts above it now seems like they are getting written to that file, is that accurate?
Ie you see more than 2 lines in that file.

The only thing in code that will ignore those requests is rain delay, make sure rain delay is off.

If lines are getting written to that file, and running said lines manually makes the zones turn on, then the problem is with now with cron.  Can you confirm all the above is accurate?

As for debug, I don’t think sprinklerd logs will help anymore, if all the above is accurate, problem is with cron.  Do you see any cron errors in your logs?  Ie /var/log/messages or where ever you have syslog set to?

woodgieboodgie, 2019-08-09:
correct, can see multiple lines in /etc/cron.d/sprinklerd. rain delay off.

sfeakes, 2019-08-09:
OK, can you try this.   Open two terminals (ssh connections) to the pi.  
In one terminal run this command
``` tail -f /var/log/messages | grep --line-buffered cron```
In the above, the file `/var/log/messages` may be `/var/log/syslog` depending on your system config and version of raspbian. so please make sure you have the right file before going any further. I don;t believe you will have both files, so do an ls and see which one you have.

In the other terminal run
```sudo sh -c 'echo "" >> /etc/cron.d/sprinklerd'```

You should see a line in the first terminal, that look like
`Aug  9 08:58:01 sprinkler cron.info cron[211]: (*system*sprinklerd) RELOAD (/etc/cron.d/sprinklerd)`
That means cron is recognizing the file and reloaded it.
If you see something like `Aug  9 08:57:01 sprinkler cron.info cron[211]: (*system*sprinklerd) ERROR (Syntax error, this crontab file will be ignored)`
That means cron didn't like something in the file.
If you see nothing at all, that means cron is either not running or is not configured with the `EXTRA_OPTS='-l'` option.

you can try
`sudo systemctl status cron`  that should state active if it's running.
The below will restart cron.
`sudo systemctl stop cron`
`sudo systemctl start cron`

Please let me know what you see in the first terminal.

woodgieboodgie, 2019-08-10:
unfortunately i dont see anything in the second window.  cron is running, the EXTRA_OPTS=''-l' is configured in /etc/default/cron

woodgieboodgie, 2019-08-10:
when i run: ps -ef | grep cron
gets the response **/usr/sbin/cron -f -l**

sfeakes, 2019-08-10:
did you try the terminal  looking at /var/log/syslog as well as /var/log/messages ?   Where cron writes will be dependent on your system configuration.

Can you post the output of this command please, see if there are any errors in the logs.
```sudo cat /var/log/syslog | grep cron```

again replace /syslog with /messages depending on your config.

woodgieboodgie, 2019-08-10:
[image]

sudo cat /var/log/syslog produces above output.  
sudo cat /var/log/messages returns nothing.

woodgieboodgie, 2019-08-10:
i changed the path to **tail -f /var/log/syslog | grep --line-buffered cron in Window1.
entered sudo sh -c 'echo "" >> /etc/cron.d/sprinklerd' in Window2.

go back to window1 and see the content that is in the screenshot above.
cant see {RELOAD} per your text above.

woodgieboodgie, 2019-08-26:
hi again.  hoping you can help me get past this issue and get the scheduler working.  What do you recommend?  if you could give me a list of things to check i am happy to do so.  i think you are correct in stating it is to do with cron, but not sure how to progress.  please help..

sfeakes, 2019-08-26:
Sorry I missed the above.
I looks like the permission of /etc/cron.d/sprinklerd has been changed some how, below command should fix that.
‘chmod 644 /etc/cron.d/sprinkled’

If the above doesn’t fix it.
Every thing else seems to be working correctly individually. There should be some form of errors in your system log, I’d suggest going through that.  Can you post the contents of /etc/cron.d/sprinklerd, just to make sure it’s getting updated correctly.

woodgieboodgie, 2019-08-27:
Ran the script above using sudo and it appears to have fixed the issue.  Shall thoroughly test and advise.

closed 2020-01-20 by woodgieboodgie

toshibochan, 2021-02-12:
I did new installation in my raspberry pi 3 but cron not up dating the schedule. I did sudo chmod 644 /etc/cron.d/sprinkled but not working. please help.

sfeakes, 2021-02-12:
Can you try the commands in the post linked below, that should at least tell you where the problem lies.

https://github.com/sfeakes/SprinklerD/issues/11#issuecomment-519934147

toshibochan, 2021-02-12:
only this have in the log. no log talking to cron:
/var/log/messages
Feb 11 21:17:33 Pi3 sprinklerd: Starting /usr/local/bin/sprinklerd version 1.1
Feb 11 21:17:33 Pi3 sprinklerd: Name = Shop Sprinkler
Feb 11 21:17:33 Pi3 sprinklerd: Port = 80
Feb 11 21:17:33 Pi3 sprinklerd: Docroot = /var/www/sprinklerd
Feb 11 21:17:33 Pi3 sprinklerd: Starting web server on port 80
Feb 11 21:17:33 Pi3 sprinklerd: MQTT client is disabled, not stating

nothing on /var/log/syslog.

cron is on and is set to EXTRA_OPTS='-l':

pi@Pi3:~ $ sudo systemctl status cron
● cron.service - Regular background program processing daemon
   Loaded: loaded (/lib/systemd/system/cron.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-02-12 00:06:26 PST; 13h ago
     Docs: man:cron(8)
 Main PID: 358 (cron)
    Tasks: 1 (limit: 1938)
   CGroup: /system.slice/cron.service
           └─358 /usr/sbin/cron -f -l

Feb 12 00:06:26 Pi3 systemd[1]: Started Regular background program processing daemon.
Feb 12 00:06:26 Pi3 cron[358]: (CRON) INFO (pidfile fd = 3)
Feb 12 00:06:26 Pi3 cron[358]: (CRON) INFO (Running @reboot jobs)

pi@Pi3:~ $ ps -ef | grep cron
root       358     1  0 12:52 ?        00:00:00 /usr/sbin/cron -f -l
pi        1502  1460  0 13:00 pts/0    00:00:00 grep --color=auto --line-buffered cron
pi        2041  1620  0 13:17 pts/2    00:00:00 grep --color=auto cron

toshibochan, 2021-02-12:
pi@Pi3:~ $ sudo cat /var/log/syslog | grep cron
Feb 12 00:06:26 Pi3 cron[358]: (CRON) INFO (pidfile fd = 3)
Feb 12 00:06:26 Pi3 cron[358]: (CRON) INFO (Running @reboot jobs)
Feb 12 13:17:01 Pi3 CRON[2021]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)

toshibochan, 2021-02-12:
sudo cat /var/log/messages returns nothing.

toshibochan, 2021-02-12:
pi@Pi3:~ $ sudo cat /var/log/syslog | grep cron
Feb 12 00:06:26 Pi3 cron[358]: (CRON) INFO (pidfile fd = 3)
Feb 12 00:06:26 Pi3 cron[358]: (CRON) INFO (Running @reboot jobs)
Feb 12 13:17:01 Pi3 CRON[2021]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Feb 12 13:26:01 Pi3 cron[358]: (*system*sprinklerd) RELOAD (/etc/cron.d/sprinklerd)
Feb 12 13:33:01 Pi3 cron[358]: (*system*sprinklerd) RELOAD (/etc/cron.d/sprinklerd)
Feb 12 13:36:01 Pi3 cron[358]: (*system*sprinklerd) RELOAD (/etc/cron.d/sprinklerd)

But no change

#***** AUTO GENERATED DO NOT EDIT *****
0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost:80?type=sensor&sensor=chanceofrain&value=0'
0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost:80?type=sensor&sensor=raintotal&value=0'
#***** AUTO GENERATED DO NOT EDIT *****

toshibochan, 2021-02-12:
/var/log/syslog

Feb 12 13:31:51 Pi3 sprinklerd: action_web_request / type=read
Feb 12 13:31:51 Pi3 sprinklerd: Request type read
Feb 12 13:31:51 Pi3 sprinklerd: WEB REQUEST read read
Feb 12 13:31:56 Pi3 sprinklerd: action_web_request / type=calcfg&day=4&zone=&time=22:00
Feb 12 13:31:56 Pi3 sprinklerd: Request type calcfg
Feb 12 13:31:56 Pi3 sprinklerd: Checking if CRON needs to be updated
Feb 12 13:31:56 Pi3 sprinklerd: Checking if CRON needs to be updated
Feb 12 13:31:56 Pi3 sprinklerd: action_web_request / type=calcfg&day=4&zone=1&time=2
Feb 12 13:31:56 Pi3 sprinklerd: Request type calcfg
Feb 12 13:31:56 Pi3 sprinklerd: Checking if CRON needs to be updated
.......
Feb 12 13:33:01 Pi3 cron[358]: (*system*sprinklerd) RELOAD (/etc/cron.d/sprinklerd)
......
but still not updating

sfeakes, 2021-02-12:
> 
> #***** AUTO GENERATED DO NOT EDIT *****
> 0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost:80?type=sensor&sensor=chanceofrain&value=0'
> 0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost:80?type=sensor&sensor=raintotal&value=0'
> #***** AUTO GENERATED DO NOT EDIT *****

This is the problem, the sprinklerd cron file is not getting updated.   Edit the settings in the web ui, wait 5 mins and cat that file, see if you get more entries.   Sprinklerd will cache a lot of stuff and write later to save CF card writes.  So it can take a few mins before that file is updated.

toshibochan, 2021-02-13:
I wait more than 5 min but nothing

sfeakes, 2021-02-13:
Can you run sprinklerd manually change the schedule and post the output.

stop sprinklerd 
sudo systemctl stop sprinklerd
run manually 
sudo sprinklerd -v -d -c /etc/sprinklerd.conf

Then set a few entries in the schedule.  You will see an error in the output if it can’t write the cron file which seems to be the case.

toshibochan, 2021-02-13:
pi@Pi3:~ $ sudo sprinklerd -v -d -c /etc/sprinklerd.conf
Notice: Starting sprinklerd version 1.1
Debug:  Config mqtt 'disabeled'
Info:   Name = Shop Sprinkler
Info:   Port = 80
Info:   Docroot = /var/www/sprinklerd
Debug:  Found 5 ZONES
Debug:  No Information for ZONE = 0
Debug:  ZONE = 1
Debug:  Zone Config        : Driveway
                      PIN : 20
         Set pull up/down : 1
                 ON state : 0
             Domoticz IDX : -1
Debug:  ZONE = 2
Debug:  Zone Config        : Parking
                      PIN : 13
         Set pull up/down : 1
                 ON state : 0
             Domoticz IDX : -1
Debug:  ZONE = 3
Debug:  Zone Config        : Genkan
                      PIN : 16
         Set pull up/down : 1
                 ON state : 0
             Domoticz IDX : -1
Debug:  ZONE = 4
Debug:  Zone Config        : Backyard
                      PIN : 12
         Set pull up/down : 1
                 ON state : 0
             Domoticz IDX : -1
Debug:  ZONE = 5
Debug:  Zone Config        : 
                      PIN : 24
         Set pull up/down : 1
                 ON state : 0
             Domoticz IDX : -1
Debug:  Found 0 INPUTS
Debug:  regexp no match (1)
Debug:  regexp no match (1)
Debug:  regexp no match (1)
Debug:  regexp no match (1)
Debug:  Read Calendar 'ON' from cache
Debug:  Read delay24h 'OFF' from cache
Debug:  Read delay24h endtime '0' from cache
Debug:  Read Chance of Rain delay '0' from cache
Debug:  Read Rain 1d delay '0.000000'in from cache
Debug:  Read Rain 2d delay '0.000000'in from cache
Debug:  Read default_runtime '1' for zone 1
Debug:  Read default_runtime '1' for zone 2
Debug:  Read default_runtime '1' for zone 3
Debug:  Read default_runtime '1' for zone 4
Debug:  Read default_runtime '10' for zone 5
Debug:  Running sprinklerd with options :- daemon=0, verbose=1, httpdport=80, debug2file=0 configfile=/etc/sprinklerd.conf
Debug:  Setting up GPIO
Debug:  Setting up Zone 1
Debug:  Set GPIO 20 to OUTPUT
Debug:  Setting up Zone 2
Debug:  Set GPIO 13 to OUTPUT
Debug:  Setting up Zone 3
Debug:  Set GPIO 16 to OUTPUT
Debug:  Setting up Zone 4
Debug:  Set GPIO 12 to OUTPUT
Debug:  Setting up Zone 5
Debug:  Set GPIO 24 to OUTPUT
Debug:  GPIO setup complete
Debug:  Starting HTTPD
Notice: Starting web server on port 80
Notice: MQTT client is disabled, not stating
Debug:  action_web_request / type=read
Debug:  Request type read
Debug:  WEB REQUEST read read
Debug:  action_web_request / type=read
Debug:  Request type read
Debug:  WEB REQUEST read read
Debug:  action_web_request / type=read
Debug:  Request type read
Debug:  WEB REQUEST read read
action_web_request / type=calcfg&day=0&zone=&time=19:20
Debug:  Request type calcfg
Debug:  Checking if CRON needs to be updated
Debug:  Checking if CRON needs to be updated
Debug:  action_web_request / type=calcfg&day=0&zone=1&time=2
Debug:  Request type calcfg
Debug:  Checking if CRON needs to be updated
Debug:  Checking if CRON needs to be updated
Debug:  Checking if CRON needs to be updated
Debug:  Checking if CRON needs to be updated
Debug:  action_web_request / type=calcfg&day=0&zone=1&time=3
Debug:  Request type calcfg
Debug:  Checking if CRON needs to be updated
Debug:  Checking if CRON needs to be updated
Debug:  Checking if CRON needs to be updated
Info:   Updating CRON
Debug:  action_web_request / type=read
Debug:  Request type read
Debug:  WEB REQUEST read read
Debug:  action_web_request / type=read
Debug:  Request type read
..............

toshibochan, 2021-02-13:
Info: Updating CRON? but not updating

toshibochan, 2021-02-13:
this is because is raspberry pi 3? only work with pi zero?

toshibochan, 2021-02-13:
[image]
is all blank

sfeakes, 2021-02-13:
Nothing to do with pi zero / 3.   Works with all Pi’s (well up to current 4).   Not enough information is being sent from browser to sprinklerd, to setup zone.  If you look in logs, zone is missing on some and start time on all.  Try using different browser, make sure you set start time, day, select zones and duration.    (Refer to image on main page, make sure it’s all filled out)

toshibochan, 2021-02-13:
@sfeakes thank you so much. Running one time in Firefox did the trick. and now work in any browser. Something in safari not working for first time. strange.

Thank you

toshibochan, 2021-02-14:
one thing I have now is zone 1  not updating on cron! zone 1 Turn on manually is working but not updating schedule.