sfeakes/SprinklerD #13: Hello.  Can you please post a basic schematic illustrating how to wire 8 relay arduino board to Pi3b?

woodgieboodgie, 2019-12-15:
I seem to have relay 8 out of commission as I have removed the jumper and wired the board to the pi as per you instructions.  Relay 8 seems to be on at all times.  What do you think I have done incorrectly?

woodgieboodgie, 2019-12-16:
I don't wish to power the Arduino 8 relay with an additional power supply and was hoping to run it off the Pi3.  Maybe mistakenly I have removed the jumper on the relay board and wired to that, which may be taking out relay number 8.  pls help

woodgieboodgie, 2019-12-29:
sfeakes.. are you still out there?  hello........

sfeakes, 2019-12-29:
It will depend on the board you have.  But, most have optical isolation and can be powered from the Pi's 5v rail.  So:-
Pi 5v         -> VCC
Pi Ground -> Ground
Pi GPIO x. -> IN #x

That should be it.
Here is a good explanation. 
https://myhydropi.com/connecting-a-relay-board-to-a-raspberry-pi

woodgieboodgie, 2019-12-30:
Thanks.  I now receive the error on boot:

"[sprinklerd] Could not register platform accessory! (Run on calendar schedule)Error: Cannot add a bridged Accessory with the same UUID as another bridged Accessory: 5d12b6af-1d24-40b5-bf8f-60e28f935854"

How do i force sprinklerd to take new UUID?

sfeakes, 2019-12-30:
The UUID is generated by Homebridge, so should be unique if all is configured correctly. I wonder if it’s you have another Homebridge accessory called “run on calendar schedule” that’s causing the problem? Or maybe you’re trying to load Homebridge-sprinklerd twice?
Other than that, if you don’t have a lot of accessories in Homebridge, you could delete the Homebridge cache and see if that helps.  But if you have a lot in Homebridge that will be a pain to reconfigure,

woodgieboodgie, 2019-12-30:
was hoping that wasnt the case, shall clear accessories and persist folders and will close this case if that works.  thanks Sean!

sfeakes, 2019-12-30:
You could always start another Homebridge (in a new directory) just to make sure all is ok, then figure out the first instance at a later date.   It’s not to hard to set a new Homebridge environment, and run two instances at the same time.  No need to install anything new, just set a few environment variables.

woodgieboodgie, 2020-01-20:
Hi again Sean.  I am still having an issue when instructing the software to cycle through the zones.  i have attached the wiring and my sprinklerd.conf for your reference.  when the arduino relay board initializes at boot the relay for zone 7 (as i am using a master valve on zone 0) is much brighter than all others.  is that contributing to the problem?  
[image]

closed 2020-01-21 by woodgieboodgie