sfeakes/SprinklerD #14: Zone 1 not working after apt-get update

toshibochan, 2020-01-18:
Zone 1 schedule not working after update the raspberry pi. I’m trying to change how long watering but always back to 0 min.

toshibochan, 2020-01-18:
This happens in 2 raspberry pi zero

sfeakes, 2020-01-19:
How are you changing the watering time (web ui / API / HomeKit)?  I assume all the other zones allow you to change watering time.  Also look in syslog see if there are any errors?

Other thing to do set set logging to file and turn on debug logging in sprinklerd config, then restart sprinklerd and look in that log for any errors.

toshibochan, 2020-01-19:
I change in web ui. I change zone one to 5 min but Later when I go back to the Schedule page the tine is back to 0. This happening only in zone one. Other zone is okay.
[image]

toshibochan, 2020-01-20:
Syslog 

Jan 19 19:51:08 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:08 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:09 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:09 DomoHouseSprinkler sprinklerd: action_web_request / type=calcfg&day=6&zone=1&time=5
Jan 19 19:51:09 DomoHouseSprinkler sprinklerd: Request type calcfg
Jan 19 19:51:09 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:09 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:10 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:10 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:11 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:11 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:12 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:12 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:13 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:13 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:13 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:13 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:14 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:14 DomoHouseSprinkler sprinklerd: action_web_request / type=read
Jan 19 19:51:14 DomoHouseSprinkler sprinklerd: Request type read
Jan 19 19:51:14 DomoHouseSprinkler sprinklerd: WEB REQUEST read read
Jan 19 19:51:14 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:14 DomoHouseSprinkler sprinklerd
Jan 19 19:51:30 DomoHouseSprinkler sprinklerd: Checking if CRON needs to be updated
Jan 19 19:51:30 DomoHouseSprinkler sprinklerd: Updating CRON
Jan 19 19:51:50 DomoHouseSprinkler sprinklerd: action_web_request / type=homebridge
Jan 19 19:51:50 DomoHouseSprinkler sprinklerd: Request type homebridge
Jan 19 19:52:01 DomoHouseSprinkler cron[235]: (*system*sprinklerd) RELOAD (/etc/cron.d/sprinklerd)

toshibochan, 2020-01-20:
In the debug log show updating but still not saving in the web ui

toshibochan, 2020-01-20:
pi@DomoHouseSprinkler:~$ sudo cat /var/log/syslog | grep cron
Jan 20 07:11:01 DomoHouseSprinkler cron[235]: (*system*sprinkl
erd) RELOAD (/etc/cron.d/sprinklerd)
pi@DomoHouseSprinkler:~$ sudo cat /var/log/mesages | grep cron
 
cat: /var/log/mesages: No such file or directory
pi@DomoHouseSprinkler:~$ sudo cat /var/log/messages | grep cro
n
pi@DomoHouseSprinkler:~$ 
pi@DomoHouseSprinkler:~$

toshibochan, 2020-01-20:
If I set zone 1 to 5 min to run now option in the schedule is set to 5 min automatically.
Is so strange behavior

toshibochan, 2020-01-20:
[image]
[image]

sfeakes, 2020-01-21:
Sounds like it’s having issues writing the cron file.  
Can you check the permissions (including directory) and time date stamp on /etc/cron.d/sprinklerd .
Should have read/write to root and time stamp should be last time you updated the schedule.

toshibochan, 2020-01-21:
Zone one not there

/etc/cron.d/sprinklerd                                     

#***** AUTO GENERATED DO NOT EDIT *****
0 6 * * 2 root /usr/bin/curl -s -o /dev/null 'localhost:80?type=cron&zone=2&runtime=4&state=on'
4 6 * * 2 root /usr/bin/curl -s -o /dev/null 'localhost:80?type=cron&zone=3&runtime=1&state=on'
0 6 * * 6 root /usr/bin/curl -s -o /dev/null 'localhost:80?type=cron&zone=2&runtime=3&state=on'
3 6 * * 6 root /usr/bin/curl -s -o /dev/null 'localhost:80?type=cron&zone=3&runtime=1&state=on'
0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost:80?type=sensor&sensor=chanceofrain&value=0'
0 0 * * * root /usr/bin/curl -s -o /dev/null 'localhost:80?type=sensor&sensor=raintotal&value=0'
#***** AUTO GENERATED DO NOT EDIT *****

sfeakes, 2020-01-21:
So it’s not updating cron file, Go back to the logs, look for error.  What are the permissions on the file? Are you using a RO root filesystem?   Are you sure other zones are updating appropriately, as that file doesn’t match the UI screen shot you posted.

toshibochan, 2020-02-02:
I add more zone and stop using zone 1. Is so strange only zone one is not working.

sfeakes, 2020-02-07:
Can you post your cfg file?

woodgieboodgie, 2020-05-06:
getting a similar issue. web server works, sprinklerd works, scheduler appears fine, but no zone 1. i am using a master valve so i have the seven zones.

woodgieboodgie, 2020-05-06:
cron writes activity of any other zone, just not zone 1. pls help.

woodgieboodgie, 2020-05-07:
[image]

manual trigger of zone 1 works

woodgieboodgie, 2020-05-07:
Error loading platform requested in your config.json at position 2
[5/7/2020, 8:08:32 PM] Error: The requested platform 'sprinklerd' was not registered by any plugin.
    at PluginManager.getPluginForPlatform (/usr/lib/node_modules/homebridge/src/pluginManager.ts:223:15)
    at /usr/lib/node_modules/homebridge/src/server.ts:386:37
    at Array.forEach (<anonymous>)
    at Server.loadPlatforms (/usr/lib/node_modules/homebridge/src/server.ts:373:27)
    at Server.start (/usr/lib/node_modules/homebridge/src/server.ts:152:29)
    at cli (/usr/lib/node_modules/homebridge/src/cli.ts:78:10)
    at Object.<anonymous> (/usr/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (internal/modules/cjs/loader.js:1133:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1153:10)
    at Module.load (internal/modules/cjs/loader.js:977:32)

woodgieboodgie, 2020-05-07:
[image]

[image]

toshibochan, 2020-05-07:
Same for me but I wasn’t able to look into of lack of time. I just add more zone and I stop using zone 1 for now.

woodgieboodgie, 2020-05-08:
Thanks, but i cant see Zone 8.  I have edited sprinklerd.conf, added an 8th zone and deleted all files in /accessories and /persist.. and restarted.  Did you modify anything else to get an additional zone working and visible through the browser?

toshibochan, 2020-05-08:
I was using only 4 zones.

closed 2020-09-27 by sfeakes