sfeakes/SprinklerD #16: version compatability

woodgieboodgie, 2020-01-24:
hi sean,

what version of node.js and npm does the plugin work with?

woodgieboodgie, 2020-02-01:
hi, can you please address the above question when you have a moment?  what versions has it been tested with?

sfeakes, 2020-02-01:
Sprinklerd has nothing to do with node.   You probably mean Homebridge-sprinklerd, if you look in package.js in that repo, it’ll tell you. Greater than 4 for node.  (Although not tested with latest node)
https://github.com/sfeakes/homebridge-sprinklerd/blob/master/package.json

Are you having a problem with installing / running Homebridge-sprinklerd

woodgieboodgie, 2020-02-01:
install is fine. u r right, its homebridge and the latest config-uix is suggesting that nodejs and npm are behind.  wary of upgrading as everything is working

sfeakes, 2020-02-01:
Don’t upgrade unless absolutely necessary.  NPM and node Can/are a real pain to upgrade at times. if it’s working leave it, there are no real security issues to worry about from this installation.  Obviously can’t talk for anything else you have running on node.

closed 2020-02-04 by woodgieboodgie