sfeakes/SprinklerD #17: cycle all zones hangs: possible homebridge issue

woodgieboodgie, 2020-02-04:
Hi Sean,

Unusual activity in log, Sprinklerd didnt get beyond Zone3.  What is rngd[398] referenced below?  must be something that is interfering with Sprinklerd because its not the first time its happened.  Funnily enough, if it doesnt occur, the 'cycle all zones'' manages to work though all zones successfully.

Feb  4 19:59:45 Homebridge sprinklerd: Turning on Zone 1
Feb  4 19:59:45 Homebridge sprinklerd: Turning on Zone 0
Feb  4 20:00:29 Homebridge vncserver-x11[426,root]: Connections: disconnected: 192.168.0.40::52980 (TCP) ([ViewerClosed] VNC Viewer closed)
Feb  4 20:00:45 Homebridge sprinklerd: Turning off Zone 1
Feb  4 20:00:45 Homebridge sprinklerd: Turning on Zone 2
Feb  4 20:01:45 Homebridge sprinklerd: Turning off Zone 2
Feb  4 20:01:45 Homebridge sprinklerd: Turning on Zone 3
Feb  4 20:04:16 Homebridge rngd[398]: stats: bits received from HRNG source: 80064
Feb  4 20:04:16 Homebridge rngd[398]: stats: bits sent to kernel pool: 34848
Feb  4 20:04:16 Homebridge rngd[398]: stats: entropy added to kernel pool: 34848
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS 140-2 successes: 4
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS 140-2 failures: 0
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Poker: 0
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Runs: 0
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Long run: 0
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Feb  4 20:04:16 Homebridge rngd[398]: stats: HRNG source speed: (min=645.960; avg=754.343; max=809.418)Kibits/s
Feb  4 20:04:16 Homebridge rngd[398]: stats: FIPS tests speed: (min=3.887; avg=5.806; max=7.112)Mibits/s
Feb  4 20:04:16 Homebridge rngd[398]: stats: Lowest ready-buffers level: 2
Feb  4 20:04:16 Homebridge rngd[398]: stats: Entropy starvations: 0
Feb  4 20:04:16 Homebridge rngd[398]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Feb  4 20:17:01 Homebridge CRON[2962]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Feb  4 21:04:16 Homebridge rngd[398]: stats: bits received from HRNG source: 120064
Feb  4 21:04:16 Homebridge rngd[398]: stats: bits sent to kernel pool: 64608
Feb  4 21:04:16 Homebridge rngd[398]: stats: entropy added to kernel pool: 64608
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS 140-2 successes: 6
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS 140-2 failures: 0
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Poker: 0
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Runs: 0
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Long run: 0
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Feb  4 21:04:16 Homebridge rngd[398]: stats: HRNG source speed: (min=645.960; avg=772.373; max=812.076)Kibits/s
Feb  4 21:04:16 Homebridge rngd[398]: stats: FIPS tests speed: (min=3.887; avg=6.174; max=7.112)Mibits/s
Feb  4 21:04:16 Homebridge rngd[398]: stats: Lowest ready-buffers level: 2
Feb  4 21:04:16 Homebridge rngd[398]: stats: Entropy starvations: 0
Feb  4 21:04:16 Homebridge rngd[398]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us

sfeakes, 2020-02-07:
`rngd` is nothing to do with SprinklerD, it's a linux random number generator.  Is SprinklerD still running, or does it die?

woodgieboodgie, 2020-02-08:
ok (am still a beginner and figuring my way around). 
Sprinklerd dies when [rngd] is doing its thing.  further testing sees 'run on calendar' and 'cycle all zones' work from start to finish if not interrupted by [rgnd] .. unless something else is interfering.
what do you suggest?

sfeakes, 2020-02-08:
Are you using a Pi4?  Quick google seems there are some problems with rngd and Pi4.  Other than that, figure out why it’s actually running, maybe just disable it. It a service and I’m not sure why it’s enabled on your machine

woodgieboodgie, 2020-02-10:
Hi, I am on a Pi3.  The service must have been installed via other plugins that I am running.  out of curiosity, how do i disable it?  and Qn2.. how could i schedule sprinklerd to automatically restart each day or multiple times per day to get around this in the short term?

woodgieboodgie, 2020-02-14:
hi sean, how do i disable that service?

closed 2020-02-22 by woodgieboodgie