sfeakes/SprinklerD #18: NODEJS 13.9

woodgieboodgie, 2020-02-22:
Once upgraded to NodeJs 13.9, sprinklerd still starts, can still switch on/off the arduino relay board zones via the GPIOs, and the 'My Sprinklers' button illuminates in Homekit.
However, the individual zone buttons themselves no longer work and if manually triggered they display 'Idle'.  this is also the case for 'run on calendar', and 'cycle all zones'.

Is there something that I need to recompile Sean or rebuild?
(had to upgrade NodeJS as other plugins needed later than 8.x)

woodgieboodgie, 2020-02-22:
subsequent testing revealed that the zone buttons do illuminate, but not real time.   when the phone running homebridge changes to a different app and then returns to homebridge the correct zone illuminates.  however, if i was to stay on homebridge when the zone changes the app doesnt update the button (unless i leave homebridge and come back again)

sfeakes, 2020-02-22:
If you start homebridge manually in debug mode do you get any errors on the console when turning zones on/off ?

woodgieboodgie, 2020-03-25:
[Config] [homebridge-sprinklerd] Failed to check registry.npmjs.org for updates: Request failed with status code 404

this is the error being received

closed 2020-04-04 by woodgieboodgie