sfeakes/SprinklerD #21: Web front end zones switch on but gui switch doesn't

DrJeff, 2020-09-24:
When I click `ON`  in the GUI the zone is turned on (pin is toggled) but the GUI shows off still. When the timer finishes there is no MQTT sent of zone turning off just timer complete is this the intended operation?

Just check with a LED on the GPIO no Power toggling on GPIO. Permissions?

sfeakes, 2020-09-24:
No that's not correct, something is up with system or config.   You should first start with a recompile though, make sure binary picks up your OS & Pi# since that's very important for GPIO.

DrJeff, 2020-09-25:
Ok thanks that makes sense now, my Pi shows up as unknown

closed 2020-09-25 by DrJeff

sfeakes, 2020-09-25:
OK.  That's the problem,  Try compiling against with the wiring_pi option.  (in the advanced make section on the first page).

I obviously need to update my GPIO library in the meantime.  What Pi do you have?

sfeakes, 2020-09-26:
Just posted update.  I figure you were using Pi4 and the GPIO control hadn't been updated for a Pi4, but it is now.