sfeakes/SprinklerD #22: "Include in status"

woodgieboodgie, 2020-09-27:
It would be very handy to see the status of the sprinklers in the banner at the top of Homekit.  Can the "My Sprinklers' button display its status or even better the individual zones?
[image]

sfeakes, 2020-09-28:
That’s a setting (Or fixed) in whatever app you are using for HomeKit on iOS device.

woodgieboodgie, 2020-09-28:
Using Homekit and there is no setting (within the individual sprinkler objects) to be able to "Include in Home Status"
[image]

closed 2021-01-06 by woodgieboodgie