sfeakes/SprinklerD #23: calendar schedule not working with Zone 1

bigjuanpa, 2020-10-21:
Hi for some reason when I use Run on calendar schedule the Zone 1 never start, the rest of the zones works fine.
Any idea how to fix or check that?

toshibochan, 2021-02-14:
same for me to zone one schedule not working

sfeakes, 2021-02-14:
On the zone 1 issue, a few people have complained, There must be a bug when you don't use a "master valve" setting. I haven't tracked that down yet. You can either start from zone 2 or what might be better is to set Zone 0 to a GPIO that's not used.

toshibochan, 2021-02-14:
putting zone 0 back not working.

sfeakes, 2021-02-14:
Just posted an update to fix the zone 1 & schedule issue.

toshibochan, 2021-02-15:
it working now! thank you so much. now I'm going to up date my oder device. you are great!!!!!!

toshibochan, 2021-02-15:
@bigjuanpa you can close this issue because is fixed in the latest update