sfeakes/SprinklerD #25: sprinklerDarkskys.sh : please change for Openweather

woodgieboodgie, 2021-01-28:
Hi Sean,

Is it possible you could edit the sprinklerDarkskys.sh to work with Openweather?  Instead of "precipProbability" they present chance of rain as "pop" and the URL is slightly different too:

https://api.openweathermap.org/data/2.5/forecast?q=Melbourne,AU&appid=123456abcd
vs
https://api.darksky.net/forecast/"$darkskyAPI"/"$location

sfeakes, 2021-01-29:
I added a new script sprinkerOpenWeather.sh
I couldn't use that specific openweatherAPI due to the probability being listed on an hourly basis, and the script / sprinklerd is designed for a daily probability. But OpenWeather do provide a different API that gives you a daily probability, so I used that.

Just a note
For my location OpenWeather probability is highly inaccurate compared to darkskys, but for your location maybe not, as I'm sure there is a reason you wanted to use OpenWeather.

woodgieboodgie, 2021-01-29:
Worked perfectly.  Thank you.

Shall keep an eye on OpenWeather accuracy, but I made the move as Darkskys are turning off at the end of the year from what I have read

sfeakes, 2021-01-30:
Crap, thanks for the heads up, I have a lot of stuff depending on their API's.  Looks like I have some time to make the move, but yes it's being shutdown.  Shame, it's been an amazing service compared to other free ones out there.

_Our API service for existing customers is not changing today, but we will no longer accept new signups. The API will continue to function through the end of 2021._

woodgieboodgie, 2021-02-03:
**Today:- Chance of rain: 11%, Rain total: 0.00"**

Above line appears in my browser when running sprinklerDarksky.sh, but not when sprinklerOpenweather.sh is run.

sfeakes, 2021-02-03:
If the probability is 0% then nothing is displayed.  My guess is OpenWeather is probability of 0, where darksky is 11.
Run the script in a terminal and you should see what it states.

closed 2021-02-23 by woodgieboodgie