sfeakes/SprinklerD #28: Master valve remains on

woodgieboodgie, 2024-09-09:
Hello,

Reinstalling sprinklerd on new Pi4, its been working fine on old pi4 for yrs.
Issue is the master valve remains on when any zone has completed its cycle. in fact it's on by default when the Pi4 is rebooted.
Schedule works properly, mqtt sees each of the topics but for zone 0 there is always a '1' against it.
Web server works fine too, can manually trigger zones.
Edited permissions on cron per other suggestion posted on this github.
Any suggestions?

[image]

sfeakes, 2024-09-09:
Can you post your config file.  Also are you using the binary included in the repo (32bit) or did you compile and are running 64 bit Linux for the Pi4 ?

woodgieboodgie, 2024-09-09:
Tried to compile straight from repo, that didnt work.. So used the 'make' option. Am running x64 on Pi4 with locally installed HB v1.8.4 and sprinklerd v0.1.2.
Via mqtt explorer when i sent a '0' to Zone:0 the master switches off, otherwise its always on from boot and when a zone switches off it also stays on.

Sprinklerd.conf below:

[SPRINKLERD]
PORT=80
NAME=Sprinklers
DOCUMENTROOT = /var/www/sprinklerd
CACHE = /var/cache/sprinklerd.cache
# The log level. [DEBUG, INFO, NOTICE, WARNING, ERROR]
#LOG_LEVEL = DEBUG
LOG_LEVEL = NOTICE

# mqtt stuff (address and topic are mandatory if you want to use MQTT)
MQTT_ADDRESS = localhost:1883
#MQTT_USER = someusername
#MQTT_PASSWD = somepassword
MQT_TOPIC = sprinklerd

# if you are using domoticz and MQTT uncomment
#MQTT_DZ_PUB_TOPIC = domoticz/in
#MQTT_DZ_SUB_TOPIC = domoticz/out

#DZIDX_CALENDAR = 197
#DZIDX_24HDELAY = 198
#DZIDX_ALL_ZONES = 199
#DZIDX_RAINSENSOR = 48

# Options for the below ZONE and GPIO configuration
#
# LOW        0
# HIGH       1
#
# PUD_OFF      0
# PUD_DOWN     1
# PUD_UP       2

#NAME           = name of zone
#GPIO_PIN       = GPIO # This is the GPIO# not the pin#, so (17 = GPIO17 = Pin 11) or another example (5 = GPIO5 = Pin 29)
#WPI_PIN        = use instead of GPIO_PIN if compiled with USE_WIRINGPI flag, This is WiringPi Pin#, not Raspberry Pi board pin#
#GPIO_PULL_UPDN = setup pull up pull down resistor  PUD_OFF|PUD_DOWN|PUD_UP
#GPIO_ON_STATE  = State GPIO reads when relay for zone is on. HIGH or LOW  1 or 0
#DOMOTICZ_IDX   = Domoticz IDX 0 or remove entry if you don;t use Domoticz (only if you use domoticz)

# Don't use ZONE:0 for anything other than master valve, if you don't have a master valve simply delete it and start from ZONE:1
[ZONE]
[ZONE:0]
NAME=Master Valve
GPIO_PIN=4
WPI_PIN=0
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0

[ZONE:1]
NAME=Lawn East
DEFAULT_RUNTIME=10
GPIO_PIN=18
WPI_PIN=1
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=200

[ZONE:2]
NAME=Lawn West
DEFAULT_RUNTIME=10
GPIO_PIN=17
WPI_PIN=2
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=201

[ZONE:3]
NAME=Garden Beds
DEFAULT_RUNTIME=10
GPIO_PIN=27
WPI_PIN=3
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=202

[ZONE:4]
NAME=Lawn North
DEFAULT_RUNTIME=10
GPIO_PIN=23
WPI_PIN=4
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=203

[ZONE:5]
NAME=Lawn South
DEFAULT_RUNTIME=10
GPIO_PIN=22
WPI_PIN=5
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=204

[ZONE:6]
NAME=Swimming Pool
DEFAULT_RUNTIME=10
GPIO_PIN=24
WPI_PIN=6
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=205

[ZONE:7]
NAME=Storm Water
DEFAULT_RUNTIME=10
GPIO_PIN=25
WPI_PIN=7
GPIO_PULL_UPDN=1
GPIO_ON_STATE=0
#DOMOTICZ_IDX=206

sfeakes, 2024-09-09:
You have master valve set to use GPIO 4, this is often used for serial, i2c & clock.  So something else might be using it.  Can you try setting it to something else eg (5,6,13,19). No need to rewire, just change the pin and restart.  If that works as expected, then move the physical wire over or look at what might be causing the conflict.  (/boot/config.txt would be a good place to start).   

I’ll compile on x64 and try on a Pi4 when I get chance and see if I can replicate.

sfeakes, 2024-09-09:
I've just compiled this for x64 and tried on Pi4 running Raspberry OS / Debian Bookworm and all worked as expected.

As above I think you have something else interfering with GPIO4. if you run `make gpio_tools` you will get some tools that may help you track this down.

woodgieboodgie, 2024-09-10:
Spot on, that did it. :)
Original build was probably x32 and a number of yrs ago and had the Master set to GPIO 4... so is root cause because the underlying changes introduced by x64?